acxz/gazebo-arch #1: Package Cleanup Thread

acxz, 2022-05-09:
This thread tracks the AUR packages which need to be imported into this repo.

We also need to discuss maintainership roles as well. This is primarily to prevent people from pushing to the same AUR package with the same change.

Essentially the AUR should only be a sync of this repo and should not be pushed directly. [`aurpublish`](https://github.com/eli-schwartz/aurpublish) will be used.

The process goes like so:
1) Commit changes to this repo
2) Use aurpublish <package_name> to publish to the AUR

List of packages to add and consolidate maintainer role

- [ ] gazebo
- [ ] gazebo-git

- [ ] sdformat
- [ ] sdformat-6
- [ ] sdformat-8
- [ ] sdformat-9
- [ ] libsdformat-hg

- [ ] ignition-cmake
- [ ] ignition-cmake-0

- [ ] ignition-common
- [ ] ignition-common-1
- [ ] ignition-common-3

- [ ] ignition-fuel_tools
- [ ] ignition-fuel_tools-1
- [ ] ignition-fuel_tools-4

- [ ] ignition-gazebo

- [ ] ignition-gui
- [ ] ignition-math
- [ ] ignition-math-4
- [ ] ignition-math-hg

- [ ] ignition-msgs
- [ ] ignition-msgs-1
- [ ] ignition-msgs-5

- [ ] ignition-physics

- [ ] ignition-plugin

- [ ] ignition-rendering

- [ ] ignition-sensors

- [ ] ignition-tools

- [ ] ignition-transport
- [ ] ignition-transport-4
- [ ] ignition-transport-8

- [ ] igniiton-utils

alexdewar, 2022-05-09:
Should I raise this as an issue btw?

alexdewar, 2022-05-09:
Okey doke

alexdewar, 2022-05-09:
Done

bobosito000, 2022-05-09:
Hi. Thanks for letting me know bout this billy. I do not mind helping. What do you guys need with the gazebo-git package?

bobosito000, 2022-05-09:
I would not mind giving you back ownership, if you use make instead of ninja on gazebo-git.

acxz, 2022-05-09:
I personally prefer make over ninja, since make respects `/etc/makepkg.conf`.

bobosito000, 2022-05-09:
Just curious, why do you use ninja instead of make in the gazebo package?

acxz, 2022-05-09:
@bobosito000 can you make a new issue about that. I'd like for this thread to stay on topic.

bobosito000, 2022-05-09:
Ok. I went ahead and added you as a maintainer.

acxz, 2022-05-09:
Sweet, verified, thanks so much @bobosito000 !

homalozoa, 2022-05-10:
Hey @acxz . I've added you into all AUR packages about ignition/sdformat I maintained. Check it please. Thanks for your helps.

acxz, 2022-05-17:
@homalozoa would you disowning those packages as well, so that I can be the primary maintainer.

homalozoa, 2022-05-19:
> @homalozoa would you disowning those packages as well, so that I can be the primary maintainer.

That's OK for me. I'll disown these packages.

acxz, 2022-06-01:
Out dated packages have been deleted from the AUR.

closed 2022-06-01 by acxz