acxz/gazebo-arch #11: Gazebo build system is not make

bobosito000, 2022-05-09:


acxz, 2022-05-09:
I also would prefer the build system to be make instead of ninja. Can you comment on this @alexdewar ? With make `/etc/makepkg.conf` is respected whereas with ninja we cannot specify the number of cores to use via an environmental flag.

alexdewar, 2022-05-10:
I like ninja personally (it's less verbose and can be faster than make) but I'm not wedded to it. I'm not sure how many people would actually want *not* to use all their cores to build a big project like gazebo, but I suppose using make does give that option.

I'm happy to change it to make if that's what people want though.

acxz, 2022-05-10:
> I'm not sure how many people would actually want not to use all their cores to build a big project like gazebo

Anyone that has a lot of cores, but limited RAM. Everyone's notion of big is different, for me gazebo is small and therefore I'm okay with ninja. But say if I wanted to build LLVM, I would run out of memory on my 64GB system with 12 cores. In such a case, I would want to limit the number of cores. ninja may be a more efficient tool, but it is inflexible. While we can work around like how the `llvm-git` package does, the burden rises for the user.

Let's change it to `make`.

alexdewar, 2022-05-10:
Good point. Let's do it.

bobosito000, 2022-05-10:
Agree with @acxz. Most of my machines are older, so having the option to change the job count is essential for me.

closed 2022-05-11 by alexdewar