acxz/gazebo-arch #13: [ignition-transport] ignition-msgs protobuf file not found

acxz, 2022-05-10:
```
[ 33%] Building CXX object src/CMakeFiles/ignition-fuel_tools7.dir/ign.cc.o
In file included from /home/acxz/vcs/git/github/acxz/gazebo-arch/ignition-fuel_tools/src/gz-fuel-tools-ignition-fuel-tools7_7.0.0/src/FuelClient.cc:22:
/usr/include/ignition/msgs8/ignition/msgs/fuel_metadata.pb.h:17:2: error: #error This file was generated by an older version of protoc which is
   17 | #error This file was generated by an older version of protoc which is
      |  ^~~~~
/usr/include/ignition/msgs8/ignition/msgs/fuel_metadata.pb.h:18:2: error: #error incompatible with your Protocol Buffer headers. Please
   18 | #error incompatible with your Protocol Buffer headers. Please
      |  ^~~~~
/usr/include/ignition/msgs8/ignition/msgs/fuel_metadata.pb.h:19:2: error: #error regenerate this file with a newer version of protoc.
   19 | #error regenerate this file with a newer version of protoc.
      |  ^~~~~
In file included from /home/acxz/vcs/git/github/acxz/gazebo-arch/ignition-fuel_tools/src/gz-fuel-tools-ignition-fuel-tools7_7.0.0/src/FuelClient.cc:22:
/usr/include/ignition/msgs8/ignition/msgs/fuel_metadata.pb.h:26:10: fatal error: google/protobuf/generated_message_table_driven.h: No such file or directory
   26 | #include <google/protobuf/generated_message_table_driven.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [src/CMakeFiles/ignition-fuel_tools7.dir/build.make:104: src/CMakeFiles/ignition-fuel_tools7.dir/FuelClient.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/acxz/vcs/git/github/acxz/gazebo-arch/ignition-fuel_tools/src/gz-fuel-tools-ignition-fuel-tools7_7.0.0/src/ign.cc:26:
/usr/include/ignition/msgs8/ignition/msgs/fuel_metadata.pb.h:17:2: error: #error This file was generated by an older version of protoc which is
   17 | #error This file was generated by an older version of protoc which is
      |  ^~~~~
/usr/include/ignition/msgs8/ignition/msgs/fuel_metadata.pb.h:18:2: error: #error incompatible with your Protocol Buffer headers. Please
   18 | #error incompatible with your Protocol Buffer headers. Please
      |  ^~~~~
/usr/include/ignition/msgs8/ignition/msgs/fuel_metadata.pb.h:19:2: error: #error regenerate this file with a newer version of protoc.
   19 | #error regenerate this file with a newer version of protoc.
      |  ^~~~~
In file included from /home/acxz/vcs/git/github/acxz/gazebo-arch/ignition-fuel_tools/src/gz-fuel-tools-ignition-fuel-tools7_7.0.0/src/ign.cc:26:
/usr/include/ignition/msgs8/ignition/msgs/fuel_metadata.pb.h:26:10: fatal error: google/protobuf/generated_message_table_driven.h: No such file or directory
   26 | #include <google/protobuf/generated_message_table_driven.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

bobosito000, 2022-05-12:
I would try this.
```
sudo pacman -U https://archive.archlinux.org/repos/2022/04/01/extra/os/x86_64/protobuf-3.19.4-1-x86_64.pkg.tar.zst
```

acxz, 2022-05-12:
I can't seem to reproduce this error anymore, after rebuilding `ignition-msgs`.

Downgrading `protobuf` should not be a recommended solution.

closed 2022-05-12 by acxz

bobosito000, 2022-05-12:
Thank you for the feedback. Someone on the aur comments suggested
downgrading protobuf after protobuf was recently updated and that worked on
my machines. Probably because I built ignition messages with the previous
version of protobuf.

I would imagine whenever protobuf updates and u want to update gazebo. U
would have to recompile all of gazebos dependencies with protobuf first
then gazebo. Right?

acxz, 2022-05-12:
Exactly

bobosito000, 2022-05-12:
That's a real pain. Is there any way to automate that process from the
gazebo pkgbuild?

acxz, 2022-05-12:
This cannot be automated inside the PKGBUILD, but rather needs to occur at the maintainer level. The proper solution is to figure out any changes in the linked `.so` files (i.e. if the package was built with `.so.1` and `.so.2` is the current version of the library installed). If this is the case, then simply bumping the `pkgrel` will suffice as that will force a rebuild.

Essentially some logic like so: https://github.com/arch4edu/arch4edu/issues/78#issuecomment-673589824

There was a more thorough solution I came across a while back, but I can't seem to recall where.

alexdewar, 2022-05-13:
> I can't seem to reproduce this error anymore, after rebuilding `ignition-msgs`.
> 
> Downgrading `protobuf` should not be a recommended solution.

This is the right way to fix this problem. Both the old *and* the new versions of protobuf should work fine, but you can't generate the headers with an old version of protobuf, then use them with the new version.

alexdewar, 2022-05-13:
> This cannot be automated inside the PKGBUILD, but rather needs to occur at the maintainer level. The proper solution is to figure out any changes in the linked `.so` files (i.e. if the package was built with `.so.1` and `.so.2` is the current version of the library installed). If this is the case, then simply bumping the `pkgrel` will suffice as that will force a rebuild.
> 
> Essentially some logic like so: [arch4edu/arch4edu#78 (comment)](https://github.com/arch4edu/arch4edu/issues/78#issuecomment-673589824)
> 
> There was a more thorough solution I came across a while back, but I can't seem to recall where.

There may or may not be changes to the ``.so`` file. The problem is to do with mismatching versions of protobuf. I'm not sure how best to automate this, but I think just rebuilding the ``ignition-msgs*`` package whenever ``protobuf`` is updated should do the trick. Obvs we wouldn't bump the ``pkgrel`` for an AUR package (the end-user should rebuild manually if needed), but if it were in ``[community]`` then that would be the thing to do.

bobosito000, 2022-05-14:
It feels like this is something that could be automatically handled from
pacman for the protobuf package with a post hook. Wouldn't you guys prefer
if when the protobuf package updates the aur dependencies are recompiled in
the required order?

alexdewar, 2022-05-17:
Yeah. It's the kind of thing you'd generally do manually on Arch. Automatically rebuilding and reinstalling packages seems a bit invasive and prone to breaking things.

bobosito000, 2022-10-11:
Maybe I'm wrong. Any thoughts?

alexdewar, 2022-10-11:
I've given my thoughts. This isn't the sort of thing you do with Arch. It's a DIY distro, so automatically rebuilding packages is not something we should be doing. If you install a new version of `protobuf`, you'll need to rebuild these packages I'm afraid. This is the standard practice for AUR packages.

acxz, 2022-10-11:
If you do run into this error, a more convenient way to resolve it is to use `yay`'s `--rebuildtree` flag. This will recursively rebuild all AUR dependencies in the dependency tree.

`yay -S ignition-fuel_tools --rebuildtree`