acxz/gazebo-arch #17: [gazebo] remove or keep gitignore for build files?

bobosito000, 2022-05-10:
Build files are not supposed to be tracked at all by git. Why would you not have them in gitignore?

alexdewar, 2022-05-11:
I don't think it terribly matters either way (as you say, there are generally only a few files in an AUR repo, so it's easy enough to keep track of which ones you want to keep or delete etc.) but ``.gitignore`` files *are* for build artifacts and the source files definitely are artifacts (as are the ``pkg`` and ``src`` folders).

``git clean`` can also remove ignored files if you want. Just run: ``git clean -fX``

acxz, 2022-05-11:
> .gitignore files are for build artifacts and the source files definitely are artifacts (as are the pkg and src folders)

:+1: just giving my own perspective on how I personally use `.gitignore`

> git clean can also remove ignored files if you want. Just run: git clean -fX

Say if I have build artifacts and personal config files (such as keys or `.vscode`) that I don't want to push, I believe `git clean -fX` will remove both right?

> Would like to know how you guys would approach that problem.

How I handle the solution to this problem is at the user level and ensure that I only use `git add -u` or `git add <specific file>`. Or at the merge level and ensure that at least one other person is looking at code before merging it. Or use `git reset`, redo the commit and force push.

Nothing wrong with how we are currently using the `.gitignore`, these are just my own two cents on how I use it in my projects.

bobosito000, 2022-05-11:
That's really interesting. I can see why you do it that way. It is very convenient.

acxz, 2023-03-21:
resolved in [d2b333d](https://github.com/acxz/gazebo-arch/commit/d2b333d25e88f4491b621bf45a2a745536210d60) via the removal of `.gitignore`

closed 2023-03-21 by acxz