acxz/gazebo-arch #21: [ignition-transport] incorrect sha256sum

bobosito000, 2022-05-12:
```
[bobosito@old-speedster ignition-transport]$ makepkg -si
==> Making package: ignition-transport 11.0.0-1 (Wed May 11 21:30:09 2022)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Found ignition-transport11_11.0.0.tar.gz
==> Validating source files with sha256sums...
    ignition-transport11_11.0.0.tar.gz ... FAILED
==> ERROR: One or more files did not pass the validity check!
[bobosito@old-speedster ignition-transport]$ 
```

bobosito000, 2022-05-12:
Would try degrading protobuf, if you have issues with protobuf.
```
sudo pacman -U https://archive.archlinux.org/repos/2022/04/01/extra/os/x86_64/protobuf-3.19.4-1-x86_64.pkg.tar.zst
```

acxz, 2022-05-12:
> Would try degrading protobuf, if you have issues with protobuf.

This should never be recommended.
Issues with protobuf concerning these packages usually means you need to rebuild the igntion-* packages to rebuild the protobuf messages with the newer protobuf method.

acxz, 2022-05-12:
Resolved with [12686cd](https://github.com/acxz/gazebo-arch/commit/12686cd57f347234315019d2e0efac7b2087a819)

feel free to comment if the issue still persists.

closed 2022-05-12 by acxz