acxz/gazebo-arch #24: [gazebo] Consolidate ``gazebo`` and ``gazebo-git`` PKGBUILDs

alexdewar, 2022-05-13:
They're more or less doing the same thing, so let's merge them together as far as possible for ease of maintenance.

acxz, 2022-05-17:
Merging the two depends on the TU's accepting the merge request. Feel free to send a merge request, but I highly doubt they will accept it as they are fundamentally different in the sense one is a release package and the other is a development/git package.

bobosito000, 2022-05-17:
Agree with @acxz.

alexdewar, 2022-05-17:
I think maybe I wasn't clear... I *know* that no TU will take the ``gazebo-git`` package, but I meant that we might as well also maintain it and we should propagate the fixes from one to the other. They should still obviously be separate PKGBUILDs, just more similar-looking ones!

acxz, 2022-05-17:
> I know that no TU will take the gazebo-git package

A merge request means that the TU will delete the `gazebo-git` package, no adopt it.

>  meant that we might as well also maintain it and we should propagate the fixes from one to the other. They should still obviously be separate PKGBUILDs, just more similar-looking ones!

Totally agree, sorry merge is kinda overloaded mb

acxz, 2022-06-17:
I believe with 12832ad, this has been accomplished. Feel free to comment on anything specific.

closed 2022-06-17 by acxz