acxz/gazebo-arch #31: [ignition rendering] BoundingBox.cc: Build failed on Manjaro

cptnapalm, 2022-06-06:
`/var/cache/private/pamac/ignition-rendering/src/gz-rendering-ignition-rendering6_6.4.0/src/BoundingBox.cc:`` In constructor ‘ignition::rendering::v6::BoundingBox::BoundingBox(ignition::rendering::v6::BoundingBox&&)’:
/var/cache/private/pamac/ignition-rendering/src/gz-rendering-ignition-rendering6_6.4.0/src/BoundingBox.cc:48:18: error: ‘exchange’ is not a member of ‘std’
   48 |   : dataPtr(std::exchange(_texture.dataPtr, nullptr))
      |                  ^~~~~~~~
make[2]: *** [src/CMakeFiles/ignition-rendering6.dir/build.make:76: src/CMakeFiles/ignition-rendering6.dir/BoundingBox.cc.o] Error 1`

acxz, 2022-06-13:
upstream issue: https://github.com/gazebosim/gz-rendering/issues/649

acxz, 2022-06-13:
patched with `[2b119f4](https://github.com/acxz/gazebo-arch/commit/2b119f4ca534ee7074caf7d91c657aa0a812a31a)`

cptnapalm, 2022-06-14:
Terrific.  Thank you so much.

acxz, 2022-06-14:
The issue will be open until the patch is not necessary. (next release)