acxz/gazebo-arch #32: Gazebo build fail

DavidHeiss, 2022-06-10:
Install gazebo with `paru -S gazebo` failed.

[image]

<details>
<summary>Log</summary>
<p>

```
/home/dh/.cache/paru/clone/gazebo/src/gazebo-gazebo11_11.10.2/gazebo/gui/building/EditorView.cc: In member function ‘virtual void gazebo::gui::EditorView::wheelEvent(QWheelEvent*)’:
/home/dh/.cache/paru/clone/gazebo/src/gazebo-gazebo11_11.10.2/gazebo/gui/building/EditorView.cc:219:23: warning: ‘QMatrix QGraphicsView::matrix() const’ is deprecated: Use transform() [-Wdeprecated-declarations]
  219 |   QMatrix mat = matrix();
      |                 ~~~~~~^~
In file included from /usr/include/qt/QtWidgets/QGraphicsView:1,
                 from /home/dh/.cache/paru/clone/gazebo/src/gazebo-gazebo11_11.10.2/gazebo/gui/qt.h:54,
                 from /home/dh/.cache/paru/clone/gazebo/src/gazebo-gazebo11_11.10.2/gazebo/gui/Conversions.hh:24,
                 from /home/dh/.cache/paru/clone/gazebo/src/gazebo-gazebo11_11.10.2/gazebo/gui/building/EditorView.cc:25:
/usr/include/qt/QtWidgets/qgraphicsview.h:169:48: note: declared here
  169 |     QT_DEPRECATED_X("Use transform()") QMatrix matrix() const;
      |                                                ^~~~~~
/home/dh/.cache/paru/clone/gazebo/src/gazebo-gazebo11_11.10.2/gazebo/gui/building/EditorView.cc:220:38: warning: ‘QPoint QWheelEvent::pos() const’ is deprecated: Use position() [-Wdeprecated-declarations]
  220 |   QPointF mousePosition = _event->pos();
      |                           ~~~~~~~~~~~^~
In file included from /usr/include/qt/QtGui/QResizeEvent:1,
                 from /home/dh/.cache/paru/clone/gazebo/src/gazebo-gazebo11_11.10.2/gazebo/gui/qt.h:34:
/usr/include/qt/QtGui/qevent.h:225:19: note: declared here
  225 |     inline QPoint pos() const { return p.toPoint(); }
      |                   ^~~
/home/dh/.cache/paru/clone/gazebo/src/gazebo-gazebo11_11.10.2/gazebo/gui/building/EditorView.cc:239:18: warning: ‘void QGraphicsView::setMatrix(const QMatrix&, bool)’ is deprecated: Use setTransform() [-Wdeprecated-declarations]
  239 |   this->setMatrix(mat);
      |   ~~~~~~~~~~~~~~~^~~~~
/usr/include/qt/QtWidgets/qgraphicsview.h:170:48: note: declared here
  170 |     QT_DEPRECATED_X("Use setTransform()") void setMatrix(const QMatrix &matrix, bool combine = false);
      |                                                ^~~~~~~~~
[ 69%] Automatic MOC for target ActuatorPlugin
[ 69%] Built target ActuatorPlugin_autogen
[ 69%] Automatic MOC for target AmbientOcclusionVisualPlugin
[ 69%] Built target AmbientOcclusionVisualPlugin_autogen
[ 69%] Building CXX object gazebo/gui/CMakeFiles/gazebo_gui.dir/building/GrabberHandle.cc.o
In file included from /usr/include/OGRE/OgreMeshSerializerImpl.h:37,
                 from /usr/include/OGRE/OgreMeshSerializer.h:34,
                 from /usr/include/OGRE/Ogre.h:75,
                 from /home/dh/.cache/paru/clone/gazebo/src/gazebo-gazebo11_11.10.2/gazebo/rendering/ogre_gazebo.h:26,
                 from /home/dh/.cache/paru/clone/gazebo/src/gazebo-gazebo11_11.10.2/gazebo/rendering/Conversions.hh:25,
                 from /home/dh/.cache/paru/clone/gazebo/src/gazebo-gazebo11_11.10.2/gazebo/rendering/Camera.hh:47,
                 from /home/dh/.cache/paru/clone/gazebo/src/gazebo-gazebo11_11.10.2/build/gazebo/rendering/rendering.hh:5,
                 from /home/dh/.cache/paru/clone/gazebo/src/gazebo-gazebo11_11.10.2/gazebo/test/ServerFixture.hh:53,
                 from /home/dh/.cache/paru/clone/gazebo/src/gazebo-gazebo11_11.10.2/gazebo/test/ServerFixture.cc:26:
/usr/include/OGRE/OgreEdgeListBuilder.h: In member function ‘bool Ogre::EdgeListBuilder::geometryLess::operator()(const Ogre::EdgeListBuilder::Geometry&, const Ogre::EdgeListBuilder::Geometry&) const’:
/usr/include/OGRE/OgreEdgeListBuilder.h:227: note: ‘-Wmisleading-indentation’ is disabled from this point onwards, since column-tracking was disabled due to the size of the code/headers
  227 |                 if (a.vertexSet < b.vertexSet) return true;
      | 
/usr/include/OGRE/OgreEdgeListBuilder.h:227: note: adding ‘-flarge-source-files’ will allow for more column-tracking support, at the expense of compilation time and memory
In file included from /home/dh/.cache/paru/clone/gazebo/src/gazebo-gazebo11_11.10.2/gazebo/sensors/SensorTypes.hh:23,
                 from /home/dh/.cache/paru/clone/gazebo/src/gazebo-gazebo11_11.10.2/gazebo/sensors/Sensor.hh:31,
                 from /home/dh/.cache/paru/clone/gazebo/src/gazebo-gazebo11_11.10.2/gazebo/sensors/AltimeterSensor.hh:25,
                 from /home/dh/.cache/paru/clone/gazebo/src/gazebo-gazebo11_11.10.2/build/gazebo/sensors/sensors.hh:2,
                 from /home/dh/.cache/paru/clone/gazebo/src/gazebo-gazebo11_11.10.2/gazebo/test/ServerFixture.hh:52,
                 from /home/dh/.cache/paru/clone/gazebo/src/gazebo-gazebo11_11.10.2/gazebo/test/ServerFixtureRecord.cc:21:
/home/dh/.cache/paru/clone/gazebo/src/gazebo-gazebo11_11.10.2/gazebo/common/EnumIface.hh:180:12: warning: ‘template<class _Category, class _Tp, class _Distance, class _Pointer, class _Reference> struct std::iterator’ is deprecated [-Wdeprecated-declarations]
  180 |     : std::iterator<std::bidirectional_iterator_tag, Enum>
      |            ^~~~~~~~
In file included from /usr/include/c++/12.1.0/string:45,
                 from /home/dh/.cache/paru/clone/gazebo/src/gazebo-gazebo11_11.10.2/gazebo/test/ServerFixtureRecord.cc:17:
/usr/include/c++/12.1.0/bits/stl_iterator_base_types.h:127:34: note: declared here
  127 |     struct _GLIBCXX17_DEPRECATED iterator
      |                                  ^~~~~~~~
[ 69%] Linking CXX executable gzserver
In file included from /usr/include/ignition/msgs5/ignition/msgs/MessageTypes.hh:119,
                 from /home/dh/.cache/paru/clone/gazebo/src/gazebo-gazebo11_11.10.2/gazebo/rendering/Visual.hh:32,
[... 392 lines cut ...]
[ 78%] Building CXX object plugins/CMakeFiles/AttachLightPlugin.dir/AttachLightPlugin.cc.o
make[1]: *** [CMakeFiles/Makefile2:6135: gazebo/gui/CMakeFiles/gazebo_gui.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 78%] Linking CXX shared library libActorPlugin.so
[ 78%] Built target ActorPlugin
[ 78%] Linking CXX shared library libActuatorPlugin.so
[ 78%] Built target ActuatorPlugin
[ 78%] Linking CXX shared library libTrackedVehiclePlugin.so
[ 78%] Built target TrackedVehiclePlugin
[ 78%] Linking CXX shared library libAmbientOcclusionVisualPlugin.so
[ 78%] Linking CXX shared library libArrangePlugin.so
[ 78%] Built target AmbientOcclusionVisualPlugin
[ 78%] Built target ArrangePlugin
[ 78%] Linking CXX shared library libArduCopterPlugin.so
[ 78%] Built target ArduCopterPlugin
[ 78%] Linking CXX shared library libAttachLightPlugin.so
[ 78%] Built target AttachLightPlugin
make: *** [Makefile:146: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
error: failed to build 'gazebo-11.10.2-3': 
error: packages failed to build: gazebo-11.10.2-3
⏎                                                                                                                  
```
</p></details>

DavidHeiss, 2022-06-10:
Got the same error with `gazebo-git`.

<details>
<summary>Log</summary>
<p>

```
nst std::__cxx11::basic_string<char>, gazebo::physics::JointState>; _KeyOfValue = std::_Select1st<std::pair<const std::__cxx11::basic_string<char>, gazebo::physics::JointState> >; _Compare = std::less<std::__cxx11::basic_string<char> >; _Alloc = std::allocator<std::pair<const std::__cxx11::basic_string<char>, gazebo::physics::JointState> >; _Link_type = std::_Rb_tree_node<std::pair<const std::__cxx11::basic_string<char>, gazebo::physics::JointState> >*]’
/usr/include/c++/12.1.0/bits/stl_tree.h:1895:47:   [ skipping 9 instantiation contexts, use -ftemplate-backtrace-limit=0 to disable ]
/usr/include/c++/12.1.0/bits/stl_tree.h:645:18:   required from ‘std::_Rb_tree_node<_Val>* std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::_M_clone_node(_Link_type, _NodeGen&) [with bool _MoveValue = false; _NodeGen = std::_Rb_tree<std::__cxx11::basic_string<char>, std::pair<const std::__cxx11::basic_string<char>, gazebo::physics::ModelState>, std::_Select1st<std::pair<const std::__cxx11::basic_string<char>, gazebo::physics::ModelState> >, std::less<std::__cxx11::basic_string<char> >, std::allocator<std::pair<const std::__cxx11::basic_string<char>, gazebo::physics::ModelState> > >::_Alloc_node; _Key = std::__cxx11::basic_string<char>; _Val = std::pair<const std::__cxx11::basic_string<char>, gazebo::physics::ModelState>; _KeyOfValue = std::_Select1st<std::pair<const std::__cxx11::basic_string<char>, gazebo::physics::ModelState> >; _Compare = std::less<std::__cxx11::basic_string<char> >; _Alloc = std::allocator<std::pair<const std::__cxx11::basic_string<char>, gazebo::physics::ModelState> >; _Link_type = std::_Rb_tree_node<std::pair<const std::__cxx11::basic_string<char>, gazebo::physics::ModelState> >*]’
/usr/include/c++/12.1.0/bits/stl_tree.h:1895:47:   required from ‘std::_Rb_tree_node<_Val>* std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::_M_copy(_Link_type, _Base_ptr, _NodeGen&) [with bool _MoveValues = false; _NodeGen = std::_Rb_tree<std::__cxx11::basic_string<char>, std::pair<const std::__cxx11::basic_string<char>, gazebo::physics::ModelState>, std::_Select1st<std::pair<const std::__cxx11::basic_string<char>, gazebo::physics::ModelState> >, std::less<std::__cxx11::basic_string<char> >, std::allocator<std::pair<const std::__cxx11::basic_string<char>, gazebo::physics::ModelState> > >::_Alloc_node; _Key = std::__cxx11::basic_string<char>; _Val = std::pair<const std::__cxx11::basic_string<char>, gazebo::physics::ModelState>; _KeyOfValue = std::_Select1st<std::pair<const std::__cxx11::basic_string<char>, gazebo::physics::ModelState> >; _Compare = std::less<std::__cxx11::basic_string<char> >; _Alloc = std::allocator<std::pair<const std::__cxx11::basic_string<char>, gazebo::physics::ModelState> >; _Link_type = std::_Rb_tree_node<std::pair<const std::__cxx11::basic_string<char>, gazebo::physics::ModelState> >*; _Base_ptr = std::_Rb_tree_node_base*]’
/usr/include/c++/12.1.0/bits/stl_tree.h:890:26:   required from ‘std::_Rb_tree_node<_Val>* std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::_M_copy(const std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>&, _NodeGen&) [with bool _MoveValues = false; _NodeGen = std::_Rb_tree<std::__cxx11::basic_string<char>, std::pair<const std::__cxx11::basic_string<char>, gazebo::physics::ModelState>, std::_Select1st<std::pair<const std::__cxx11::basic_string<char>, gazebo::physics::ModelState> >, std::less<std::__cxx11::basic_string<char> >, std::allocator<std::pair<const std::__cxx11::basic_string<char>, gazebo::physics::ModelState> > >::_Alloc_node; _Key = std::__cxx11::basic_string<char>; _Val = std::pair<const std::__cxx11::basic_string<char>, gazebo::physics::ModelState>; _KeyOfValue = std::_Select1st<std::pair<const std::__cxx11::basic_string<char>, gazebo::physics::ModelState> >; _Compare = std::less<std::__cxx11::basic_string<char> >; _Alloc = std::allocator<std::pair<const std::__cxx11::basic_string<char>, gazebo::physics::ModelState> >; _Link_type = std::_Rb_tree_node<std::pair<const std::__cxx11::basic_string<char>, gazebo::physics::ModelState> >*]’
/usr/include/c++/12.1.0/bits/stl_tree.h:901:29:   required from ‘std::_Rb_tree_node<_Val>* std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::_M_copy(const std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>&) [with _Key = std::__cxx11::basic_string<char>; _Val = std::pair<const std::__cxx11::basic_string<char>, gazebo::physics::ModelState>; _KeyOfValue = std::_Select1st<std::pair<const std::__cxx11::basic_string<char>, gazebo::physics::ModelState> >; _Compare = std::less<std::__cxx11::basic_string<char> >; _Alloc = std::allocator<std::pair<const std::__cxx11::basic_string<char>, gazebo::physics::ModelState> >; _Link_type = std::_Rb_tree_node<std::pair<const std::__cxx11::basic_string<char>, gazebo::physics::ModelState> >*]’
/usr/include/c++/12.1.0/bits/stl_tree.h:939:23:   required from ‘std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::_Rb_tree(const std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>&) [with _Key = std::__cxx11::basic_string<char>; _Val = std::pair<const std::__cxx11::basic_string<char>, gazebo::physics::ModelState>; _KeyOfValue = std::_Select1st<std::pair<const std::__cxx11::basic_string<char>, gazebo::physics::ModelState> >; _Compare = std::less<std::__cxx11::basic_string<char> >; _Alloc = std::allocator<std::pair<const std::__cxx11::basic_string<char>, gazebo::physics::Mo [... cut ...]
[ 78%] Linking CXX shared library libArrangePlugin.so
[ 78%] Built target ActuatorPlugin
[ 78%] Built target ArrangePlugin
[ 78%] Linking CXX shared library libArduCopterPlugin.so
[ 78%] Built target ArduCopterPlugin
[ 78%] Linking CXX shared library libAmbientOcclusionVisualPlugin.so
[ 78%] Built target AmbientOcclusionVisualPlugin
[ 78%] Linking CXX shared library libAttachLightPlugin.so
[ 78%] Built target AttachLightPlugin
make: *** [Makefile:146: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
```
</p>
</details>

acxz, 2022-06-13:
Looks like an upstream issue.
Can you make an upstream issue and link it back here so that upstream folks know about it?

acxz, 2022-06-13:
See the following relevant comment: https://github.com/osrf/gazebo/pull/3228#issuecomment-1153583441

Looks like it is best for us to just use a previous graphviz version until graphviz 4.0.1 comes out. We can temporarily create our own graphviz-3 aur package that gazebo can point to.

ruturajsambhusvt, 2022-07-09:
Hello, I am new to ArchLinux and have been trying to install a gazebo. I get the same error. I didn't really understand what the solution is after looking at the discussion. It would be great if maintainers can advise a solution.

acxz, 2022-07-09:
@alexdewar can you sync `gazebo` with the `aurpublish gazebo` command?

@ruturajsambhusvt while you wait for us to update `gazebo`, you can use the `gazebo-git` package which does have the fix. I first recommend you use an AUR helper such as `paru` or `yay`. And then you can do `paru -Syu gazebo-git`. Let me know if you still have issues!

traversaro, 2022-07-10:
By the way, graphviz 5 was also recently released upstream and once it is packaged in Arch it should solve this problem as well.

ruturajsambhusvt, 2022-07-11:
Hello, Thank you very much for your prompt help. I tried using paru -Syu gazebo-git. It tried to build for about 4 hours and threw the following error - 
error: packages failed to build: gazebo-git-r32519.e2849f96ac-1 linux-rt-5.17.1.17.realtime2-3 (linux-rt linux-rt-d
ocs linux-rt-headers).
To give some context, I need gazebo to successfully install ROS package for the Franka Emika Panda robot arm on Arch Linux realtime kernel.

acxz, 2022-07-11:
Can you try just `paru -S gazebo-git` and output the build log? You can do `paru -S gazebo-git |& tee out.txt` and post the output of `out.txt` here.

ruturajsambhusvt, 2022-07-11:
Thank you. Looks like paru -S gazebo-git |& tee out.txt worked.

Output of out.txt

:: Resolving dependencies...
:: Calculating conflicts...
:: Calculating inner conflicts...
Aur (1) gazebo-git-r32519.e2849f96ac-1
:: Proceed to review? [Y/n]:
:: Downloading PKGBUILDs...
 PKGBUILDs up to date
 nothing new to review
fetching devel info...
==> Making package: gazebo-git r32519.e2849f96ac-1 (Mon Jul 11 11:56:05 2022)
==> Retrieving sources...
  -> Updating gazebo git repo...
==> Validating source files with sha256sums...
    gazebo ... Skipped
==> Making package: gazebo-git r32519.e2849f96ac-1 (Mon Jul 11 11:56:06 2022)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Updating gazebo git repo...
==> Validating source files with sha256sums...
    gazebo ... Skipped
==> Removing existing $srcdir/ directory...
==> Extracting sources...
  -> Creating working copy of gazebo git repo...
Cloning into 'gazebo'...
done.
==> Starting prepare()...
==> Starting pkgver()...
==> Updated version: gazebo-git r32521.3e3e019f72-1
==> Sources are ready.
gazebo-git-r32519.e2849f96ac-1: parsing pkg list...
:: gazebo-git-r32519.e2849f96ac-1 is up to date -- skipping build
loading packages...
resolving dependencies...
looking for conflicting packages...
Packages (1) gazebo-git-r32521.3e3e019f72-1
Total Installed Size:  133.61 MiB
:: Proceed with installation? [Y/n]
checking keyring...
checking package integrity...
loading package files...
checking for file conflicts...
checking available disk space...
:: Processing package changes...
installing gazebo-git...
==> Now you need to export some Gazebo variables:
==> If you're using bash:
==>   echo 'source /usr/share/gazebo/setup.sh' >> ~/.bashrc
==>   source ~/.bashrc
Optional dependencies for gazebo-git
    bullet: Bullet support
    cegui: Design custom graphical interfaces
    ffmpeg: Playback movies on textured surfaces [installed]
    gdal: Digital elevation terrains support [installed]
    libdart: DART support
    libspnav: space navigator joystick support
    libusb: USB peripherals support [installed]
    simbody: Simbody support
    urdfdom: Load URDF files [installed]
:: Running post-transaction hooks...
(1/1) Arming ConditionNeedsUpdate...

alexdewar, 2022-07-11:
Glad you got it working!

Adding ``| tee out.txt`` to the command won't have made it work though. I suspect that it's just that you're now using the updated PKGBUILD so the error is no longer present.

cybaol, 2022-08-03:
modify `sed -i 's/putstr/printf/g' gazebo/gui/qgv/private/QGVCore.h` to `sed -i 's/printf/putstr/g' gazebo/gui/qgv/private/QGVCore.h` in PKGBUILD.

acxz, 2022-08-03:
@alexdewar `gazebo` is failing once again, can you remove the sed command so that this issue can be closed?

acxz, 2022-08-17:
reopening until the package is synced to the aur.
@alexdewar from this directory, you should be able to run a simple `aurpublish gazebo` and the package will be synced

alexdewar, 2022-08-17:
I ended up having to merge the AUR repo back into this one with ``aurpublish -p gazebo`` before I could push again. Not sure why :shrug: 

Done now though!

PR #37 (2022-08-17): upgpkg: gazebo 11.11.0-2
upgpkg: gazebo 11.11.0-2

remove graphvizv4 patch

closes #32
closed 2022-08-17 by alexdewar