acxz/gazebo-arch #41: ignition-rendering fails to build

doctorcolossus, 2022-10-13:
Trying to install `ignition-gazebo` fails for me while attempting to build dependency `ignition-rendering`.

There are several warnings, but it isn't clear for me what the error is.

I have an onboard Intel GPU, so no CUDA. It should still work without that, right?

Can anyone help? I need to get this installed urgently.

```
$ yay ignition-gazebo
1 aur/ignition-gazebo 6.12.0-1 (+0 0.00)
    Open source robotics simulator.
==> Packages to install (eg: 1 2 3, 1-3 or ^4)
==> 1
:: Checking for conflicts...
:: Checking for inner conflicts...
[Aur:7]  ignition-rendering-6.5.0-1  ignition-msgs-8.6.0-1  ignition-transport-11.2.0-1  ignition-gui-6.6.1-1  ignition-fuel_tools-7.1.0-1  ignition-sensors-6.6.0-1  ignition-gazebo-6.12.0-1

  7 ignition-rendering                       (Build Files Exist)
  6 ignition-msgs                            (Build Files Exist)
  5 ignition-transport                       (Build Files Exist)
  4 ignition-gui                             (Build Files Exist)
  3 ignition-fuel_tools                      (Build Files Exist)
  2 ignition-sensors                         (Build Files Exist)
  1 ignition-gazebo                          (Build Files Exist)
==> Packages to cleanBuild?
==> [N]one [A]ll [Ab]ort [I]nstalled [No]tInstalled or (1 2 3, 1-3, ^4)
==> a
:: Deleting (1/7): /home/casey/.cache/yay/ignition-rendering
:: Deleting (2/7): /home/casey/.cache/yay/ignition-msgs
:: Deleting (3/7): /home/casey/.cache/yay/ignition-transport
:: Deleting (4/7): /home/casey/.cache/yay/ignition-gui
:: Deleting (5/7): /home/casey/.cache/yay/ignition-fuel_tools
:: Deleting (6/7): /home/casey/.cache/yay/ignition-sensors
:: Deleting (7/7): /home/casey/.cache/yay/ignition-gazebo
:: (1/7) Downloaded PKGBUILD: ignition-gui
:: (2/7) Downloaded PKGBUILD: ignition-transport
:: (3/7) Downloaded PKGBUILD: ignition-msgs
:: (4/7) Downloaded PKGBUILD: ignition-fuel_tools
:: (5/7) Downloaded PKGBUILD: ignition-sensors
:: (6/7) Downloaded PKGBUILD: ignition-gazebo
:: (7/7) Downloaded PKGBUILD: ignition-rendering
  7 ignition-rendering                       (Build Files Exist)
  6 ignition-msgs                            (Build Files Exist)
  5 ignition-transport                       (Build Files Exist)
  4 ignition-gui                             (Build Files Exist)
  3 ignition-fuel_tools                      (Build Files Exist)
  2 ignition-sensors                         (Build Files Exist)
  1 ignition-gazebo                          (Build Files Exist)
==> Diffs to show?
==> [N]one [A]ll [Ab]ort [I]nstalled [No]tInstalled or (1 2 3, 1-3, ^4)
==> n
:: (1/7) Parsing SRCINFO: ignition-rendering
:: (2/7) Parsing SRCINFO: ignition-msgs
:: (3/7) Parsing SRCINFO: ignition-transport
:: (4/7) Parsing SRCINFO: ignition-gui
:: (5/7) Parsing SRCINFO: ignition-fuel_tools
:: (6/7) Parsing SRCINFO: ignition-sensors
:: (7/7) Parsing SRCINFO: ignition-gazebo
==> Making package: ignition-gui 6.6.1-1 (2022 m. spalio 13 d. 22:11:50)
==> Retrieving sources...
==> Making package: ignition-sensors 6.6.0-1 (2022 m. spalio 13 d. 22:11:50)
==> Retrieving sources...
  -> Downloading ignition-gui-6.6.1.tar.gz...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0  -> Downloading ignition-sensors-6.6.0.tar.gz...
==> Making package: ignition-gazebo 6.12.0-1 (2022 m. spalio 13 d. 22:11:50)
==> Making package: ignition-rendering 6.5.0-1 (2022 m. spalio 13 d. 22:11:50)
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
[... 161 lines cut ...]
Call Stack (most recent call first):
  CMakeLists.txt:170 (ign_configure_build)

CMake Error at /usr/share/cmake/ignition-cmake2/cmake2/IgnConfigureBuild.cmake:61 (message):
  -- 	Missing dependency [ignition-common4] (Components: graphics, events)
Call Stack (most recent call first):
  CMakeLists.txt:170 (ign_configure_build)

CMake Error at /usr/share/cmake/ignition-cmake2/cmake2/IgnConfigureBuild.cmake:63 (message):
  -- END BUILD ERRORS

Call Stack (most recent call first):
  CMakeLists.txt:170 (ign_configure_build)

CMake Error at /usr/share/cmake/ignition-cmake2/cmake2/IgnConfigureBuild.cmake:68 (message):
  Errors encountered in build.  Please see BUILD ERRORS above.
Call Stack (most recent call first):
  CMakeLists.txt:170 (ign_configure_build)

-- Configuring incomplete, errors occurred!
See also "/home/casey/.cache/yay/ignition-rendering/src/gz-rendering-ignition-rendering6_6.5.0/build/CMakeFiles/CMakeOutput.log".
==> ERROR: A failure occurred in build().
    Aborting...
 -> error making: ignition-rendering
```

acxz, 2022-10-13:
> CMake Error at /usr/share/cmake/ignition-cmake2/cmake2/IgnConfigureBuild.cmake:61 (message):
  -- 	Missing dependency [ignition-common4] (Components: graphics, events)

What's your installed `ignition-common` version?
`yay -Qs ignition-common`
If you don't have it installed, try `yay -S ignition-common`. You may have to remove `ignition-common-3` via `yay -Rns ignition-common-3`.

doctorcolossus, 2022-10-13:
Thanks for your quick reply.

I had `ignition-common-3` (from arch4edu I think) installed during my first few attempts, then I uninstalled it, seeing that `ignition-common` was listed as the main dependency with `ignition-common-3` as an alternative, and assuming that the former would get installed automatically as a dependency, but actually I don't see that happening and now I apparently have neither installed:
```
$  yay -Qs ignition-common
local/ignition-common-3 3.14.1-1
    A collection of useful classes and functions for handling many command tasks. This includes parsing 3D mesh files, managing console
    output, and using PID controllers.
```
Let me try manually pre-installing that, and then trying again to build `ignition-rendering`...

Okay, so I installed `ignition-common` (4.5.2-1) and it now `ignition-rendering` does configure correctly and is currently building...

Hopefully it succeeds and then I can get the whole `ignition-gazebo` installed, but I guess that resolves this issue! Not sure why `yay` didn't try to resolve that dependency though... 🤔 Thanks again for the quick help! If I run into any more problems, I'll open a new issue.

doctorcolossus, 2022-10-13:
Maybe `ignition-common-3` should not be listed as an alternative though.

I'm pretty sure my cmake configuration failed similarly when I tried with that one.

acxz, 2022-10-13:
Yeah the proper solution to this would be to specify `ignition-common>=4` in the `makedepends` for `ignition-gazebo`. Feel free to submit a PR about it, in the meanwhile I'll keep this issue open so its more visible if others encounter this issue.

goekce, 2023-02-13:
I had a similar error:
```
Missing dependency [ignition-common5] (Components: graphics, ...
```
Problem: `gdal` and `assimp` were missing from `ignition-common` so graphics components were not built.

acxz, 2023-02-28:
This should be resolved with 38a7924

closed 2023-02-28 by acxz