acxz/gazebo-arch #43: Segfault running ign gazebo

lounick, 2022-10-18:
Hi,

I just installed ignition-gazebo using paru. All seem to have compiled/installed normally. Trying to run `ign gazebo` though gives me a segmentation fault related to libQt5, I don't know if it is related to the error on the top about the QML file, or that I am running wayland/sway without a DE. 

My experience in Arch is rusty given that I haven't used it since 2009. So I may as well be the error in the whole process :P

Has anyone experienced something like that? 

```
[niko@niko-hpbook ~]$ ign gazebo 
[GUI] [Err] [Dialog.cc:53] Internal error: Failed to instantiate QML file [qrc:qml/StandaloneDialog.qml]
Stack trace (most recent call last):
#28   Object "[0xffffffffffffffff]", at 0xffffffffffffffff, in 
#27   Object "ign gazebo gui", at 0x55e5beead0c4, in _start
#26   Object "/usr/lib/libc.so.6", at 0x7f601a554349, in __libc_start_main
#25   Object "/usr/lib/libc.so.6", at 0x7f601a55428f, in 
#24   Object "ign gazebo gui", at 0x55e5beead084, in 
#23   Object "/usr/lib/libruby.so.3.0", at 0x7f601a8b22ea, in ruby_run_node
#22   Object "/usr/lib/libruby.so.3.0", at 0x7f601a8ae844, in 
#21   Object "/usr/lib/libruby.so.3.0", at 0x7f601aa4bd8b, in rb_vm_exec
#20   Object "/usr/lib/libruby.so.3.0", at 0x7f601aa4765e, in 
#19   Object "/usr/lib/libruby.so.3.0", at 0x7f601aa44869, in 
#18   Object "/usr/lib/libruby.so.3.0", at 0x7f601aa3b5bc, in 
#17   Object "/usr/lib/libruby.so.3.0", at 0x7f601a98a6a7, in 
#16   Object "/usr/lib/libruby.so.3.0", at 0x7f601a8affcb, in rb_protect
#15   Object "/usr/lib/libruby.so.3.0", at 0x7f601aa501d1, in rb_yield
#14   Object "/usr/lib/libruby.so.3.0", at 0x7f601aa4bd8b, in rb_vm_exec
#13   Object "/usr/lib/libruby.so.3.0", at 0x7f601aa4765e, in 
#12   Object "/usr/lib/libruby.so.3.0", at 0x7f601aa44869, in 
#11   Object "/usr/lib/libruby.so.3.0", at 0x7f601aa3b5bc, in 
#10   Object "/usr/lib/ruby/3.0.0/x86_64-linux/fiddle.so", at 0x7f601abf4acc, in 
#9    Object "/usr/lib/libruby.so.3.0", at 0x7f601aa0e135, in rb_nogvl
#8    Object "/usr/lib/ruby/3.0.0/x86_64-linux/fiddle.so", at 0x7f601abf43bc, in 
#7    Object "/usr/lib/libffi.so.8", at 0x7f601abacb72, in ffi_call
#6    Object "/usr/lib/libffi.so.8", at 0x7f601aba9f5d, in 
#5    Object "/usr/lib/libffi.so.8", at 0x7f601abad4f5, in 
#4    Object "/usr/lib/libignition-gazebo6-ign.so.6.12.0", at 0x7f60161a695d, in runGui
#3    Object "/usr/lib/libignition-gazebo6-gui.so.6", at 0x7f6015ee8a52, in ignition::gazebo::v6::gui::runGui(int&, char**, char const*, char const*, int, char const*)
#2    Object "/usr/lib/libignition-gazebo6-gui.so.6", at 0x7f6015ee7578, in ignition::gazebo::v6::gui::createGui(int&, char**, char const*, char const*, bool, char const*, int, char const*)
#1    Object "/usr/lib/libignition-gazebo6-gui.so.6", at 0x7f6015ee62f2, in ignition::gazebo::v6::gui::launchQuickStart(int&, char**, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
#0    Object "/usr/lib/libQt5Gui.so.5", at 0x7f6012d44d56, in QWindow::resize(QSize const&)
Segmentation fault (Address not mapped to object [0x8])
Escalating to SIGKILL on [Ignition Gazebo Server]
```

goekce, 2023-02-13:
I have versioni 7.3 and had a similar error:
```
$ ign sim -v        
[Msg] Gazebo Sim GUI    v7.3.0
[Msg] Gazebo Sim Quick start dialog
[Msg] Received world [https://fuel.gazebosim.org/1.0/OpenRobotics/worlds/NAO joint control] from the GUI.
[Msg] Gazebo Sim Server v7.3.0
[Wrn] [LocalCache.cc:165] Server directory does not exist [/home/u/.gz/fuel/fuel.gazebosim.org]
[Msg] Downloading world [fuel.gazebosim.org/openrobotics/worlds/nao joint control]
[GUI] [Wrn] [Application.cc:845] [QT] QQmlApplicationEngine failed to load component
[GUI] [Wrn] [Application.cc:845] [QT] qrc:qml/Main.qml:329:3: Type GzSnackBar unavailable
[GUI] [Wrn] [Application.cc:845] [QT] qrc:/qml/GzSnackBar.qml:18:1: module "QtGraphicalEffects" is not installed
[GUI] [Err] [MainWindow.cc:116] Internal error: Failed to instantiate QML file [qrc:qml/Main.qml]
[GUI] [Err] [Application.cc:152] Failed to initialize main window.
Stack trace (most recent call last):
#28   Object "[0xffffffffffffffff]", at 0xffffffffffffffff, in
...
```

`ignition-gazebo` package is missing the following deps:
```
qt5-graphicaleffects qt5-quickcontrols
```

acxz, 2023-03-02:
With the update to `7.4.0` there should be no seg fault on the `gz sim` initial command.

Closing this issue, if it still pops up, please open another issue.

closed 2023-03-02 by acxz