acxz/gazebo-arch #45: [ignition-math] Missing dependency gz-utils2

Bidski, 2023-01-21:
Trying to install `ignition-math==7.1.0` and I get this error from cmake
```
==> Making package: ignition-math 7.1.0-1 (Sat 21 Jan 2023 14:33:11)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Found ignition-math-7.1.0.tar.gz
==> Validating source files with sha256sums...
    ignition-math-7.1.0.tar.gz ... Passed
==> Removing existing $srcdir/ directory...
==> Extracting sources...
  -> Extracting ignition-math-7.1.0.tar.gz with bsdtar
==> Sources are ready.
==> Making package: ignition-math 7.1.0-1 (Sat 21 Jan 2023 14:33:14)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> WARNING: Using existing $srcdir/ tree
==> Starting build()...
-- The C compiler identification is GNU 12.2.1
-- The CXX compiler identification is GNU 12.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- gz-math7 version 7.1.0
-- Operating system is Linux
CMake Warning at /usr/share/cmake/gz-cmake3/cmake3/GzUtils.cmake:231 (find_package):
  By not providing "Findgz-utils2.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "gz-utils2", but CMake did not find one.

  Could not find a package configuration file provided by "gz-utils2" with
  any of the following names:

    gz-utils2Config.cmake
    gz-utils2-config.cmake

  Add the installation prefix of "gz-utils2" to CMAKE_PREFIX_PATH or set
  "gz-utils2_DIR" to a directory containing one of the above files.  If
  "gz-utils2" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  CMakeLists.txt:53 (gz_find_package)

-- Looking for gz-utils2 - not found

-- Looking for eigen3 - found

-- Searching for swig - found version 4.0.2.
-- Searching for Ruby - found version 3.0.5.
-- Searching for Python - found version 3.10.9.
-- Performing Test HAS_FLTO
-- Performing Test HAS_FLTO - Success
-- Searching for pybind11 - found version 2.10.3.
CMake Error at /usr/share/cmake/gz-cmake3/cmake3/GzConfigureBuild.cmake:73 (message):
  -- BUILD ERRORS: These must be resolved before compiling.
Call Stack (most recent call first):
  CMakeLists.txt:124 (gz_configure_build)

CMake Error at /usr/share/cmake/gz-cmake3/cmake3/GzConfigureBuild.cmake:75 (message):
  -- 	Missing dependency [gz-utils2]
Call Stack (most recent call first):
  CMakeLists.txt:124 (gz_configure_build)

CMake Error at /usr/share/cmake/gz-cmake3/cmake3/GzConfigureBuild.cmake:77 (message):
  -- END BUILD ERRORS

Call Stack (most recent call first):
  CMakeLists.txt:124 (gz_configure_build)

CMake Error at /usr/share/cmake/gz-cmake3/cmake3/GzConfigureBuild.cmake:82 (message):
  Errors encountered in build.  Please see BUILD ERRORS above.
Call Stack (most recent call first):
  CMakeLists.txt:124 (gz_configure_build)

-- Configuring incomplete, errors occurred!
See also "/home/bidski/.cache/yay/ignition-math/src/gz-math-gz-math7_7.1.0/build/CMakeFiles/CMakeOutput.log".
```

This PKGBUILD script installs the `gz-utils` package which fixes this issue
```
_pkgname=gz-utils
pkgname="${_pkgname}2"
pkgver=2.0.0
pkgrel=1
pkgdesc="General purpose math library for robot applications."
arch=('i686' 'x86_64')
url="https://gazebosim.org/libs/utils"
license=('Apache')
groups=('development')
depends=('gcc-libs')
source=("${pkgname}-${pkgver}.tar.gz::https://github.com/gazebosim/${_pkgname}/archive/refs/tags/${pkgname}_${pkgver}.tar.gz")
sha256sums=("af9e5b862e10aa0cedd97d9c5ca3eb9a443b7c9e560a083e8f0399e93e1cfafa")

_dir="${_pkgname}-${pkgname}_${pkgver}"

build() {
  cd "${srcdir}/${_dir}"

  mkdir -p build
  cd build

  cmake .. -Wno-dev \
           -DCMAKE_BUILD_TYPE="Release" \
           -DCMAKE_INSTALL_PREFIX="/usr" \
           -DCMAKE_INSTALL_LIBDIR="lib" \
           -DBUILD_TESTING:BOOL=False

  make
}

package() {
  cd "${srcdir}/${_dir}/build"
  make DESTDIR="${pkgdir}/" install
}
```

Cryon1x, 2023-01-21:
Installing ignition-utils also fixes the issue.

Bidski, 2023-01-22:
> Installing ignition-utils also fixes the issue.

That should probably be added as a dependency for `ignition-math` then

petronny, 2023-01-24:
Also confirm this issue in https://github.com/arch4edu/cactus/actions/runs/3986584093/jobs/6835334998

goekce, 2023-02-13:
I tried to build `ignition-math` in a chroot. The following deps were missing:

```
'python' 'pybind11' 'ignition-utils'
```

fbartelt, 2023-02-19:
I tried both workarounds mentioned and neither worked.

Using @Bidski PKGBUILD I get this error:
```
 ==> Starting build()...
-- The C compiler identification is GNU 12.2.1
-- The CXX compiler identification is GNU 12.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /sbin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /sbin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- gz-math7 version 7.1.0
-- Operating system is Linux
-- Looking for gz-utils2 -- found version 2.0.0
-- Searching for dependencies of gz-utils2
-- Looking for gz-utils2 - found

-- Looking for eigen3 - found

-- Searching for swig - found version 4.0.2.
-- Searching for Ruby - found version 3.0.5.
-- Searching for Python - found version 3.10.9.
-- Performing Test HAS_FLTO
-- Performing Test HAS_FLTO - Success
-- Searching for pybind11 - found version 2.10.3.
-- 
-- Searching for host SSE information
-- SSE2 found
-- SSE3 found
-- SSE4.1 found
-- SSE4.2 found
-- Configuring library: gz-math7
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY - Success
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY - Success
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Success
-- Testing is disabled -- skipping UNIT tests
-- Testing is disabled -- skipping UNIT tests
-- Building pybind11 interfaces
-- Configuring library: gz-math7-eigen3
-- Testing is disabled -- skipping UNIT tests
-- Adding codecheck target
-- Build configuration successful
-- Build type: Release
-- Install prefix: /usr
-- Looking for ronn to generate manpages - found
-- Found Doxygen: /sbin/doxygen (found version "1.9.6") found components: doxygen dot 
-- Configuring done
CMake Error in src/python_pybind11/CMakeLists.txt:
  Imported target "pybind11::headers" includes non-existent path

    "/include"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.
==> ERROR: A failure occurred in build().
    Aborting...
 -> error making: ignition-math
```
After installing `ignition-utils` I get
```
==> Starting build()...
-- The C compiler identification is GNU 12.2.1
-- The CXX compiler identification is GNU 12.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /sbin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /sbin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- gz-math7 version 7.1.0
-- Operating system is Linux
-- Looking for gz-utils2 -- found version 2.0.0
-- Searching for dependencies of gz-utils2
CMake Error at /lib/cmake/gz-utils2/gz-utils2-config.cmake:105 (message):
  File or directory //include/gz/utils2 referenced by variable
  gz-utils2_INCLUDE_DIRS does not exist !
Call Stack (most recent call first):
  /lib/cmake/gz-utils2/gz-utils2-config.cmake:164 (set_and_check)
  /usr/share/cmake/gz-cmake3/cmake3/GzUtils.cmake:231 (find_package)
  CMakeLists.txt:53 (gz_find_package)

-- Configuring incomplete, errors occurred!
See also "/home/fbartelt/.cache/yay/ignition-math/src/gz-math-gz-math7_7.1.0/build/CMakeFiles/CMakeOutput.log".
==> ERROR: A failure occurred in build().
    Aborting...
 -> error making: ignition-math
```
Am I missing something?

Bidski, 2023-02-19:
I was apparently missing a couple of dependencies for `gz-utils` which I already had installed. Try this 
```
_pkgname=gz-utils
pkgname="${_pkgname}2"
pkgver=2.0.0
pkgrel=1
pkgdesc="General purpose math library for robot applications."
arch=('i686' 'x86_64')
url="https://gazebosim.org/libs/utils"
license=('Apache')
groups=('development')
depends=('gcc-libs' 'pybind11' 'python' 'ignition-cmake')
source=("${pkgname}-${pkgver}.tar.gz::https://github.com/gazebosim/${_pkgname}/archive/refs/tags/${pkgname}_${pkgver}.tar.gz")
sha256sums=("af9e5b862e10aa0cedd97d9c5ca3eb9a443b7c9e560a083e8f0399e93e1cfafa")

_dir="${_pkgname}-${pkgname}_${pkgver}"

build() {
  cd "${srcdir}/${_dir}"

  mkdir -p build
  cd build

  cmake .. -Wno-dev \
           -DCMAKE_BUILD_TYPE="Release" \
           -DCMAKE_INSTALL_PREFIX="/usr" \
           -DCMAKE_INSTALL_LIBDIR="lib" \
           -DBUILD_TESTING:BOOL=False

  make
}

package() {
  cd "${srcdir}/${_dir}/build"
  make DESTDIR="${pkgdir}/" install
}
```

fbartelt, 2023-02-20:
> I was apparently missing a couple of dependencies for `gz-utils` which I already had installed. Try this

Still get the same error when installing `ignition-math`. But I solved this issue somehow. I downloaded the PKGBUILD from AUR, didn't change anything, and ran `makepkg`, got no `cmake` errors and sucessfully installed the package. 

This only worked using your PKGBUILD for gz-utils. When I tried the same thing but using `ignition-utils` from AUR I still got the same `cmake` errors.

acxz, 2023-02-28:
> I tried to build `ignition-math` in a chroot. The following deps were missing:
> 
> ```
> 'python' 'pybind11' 'ignition-utils'
> ```

Thanks for this @goekce!
Updated the PKGBUILD and SRCINFO with this info.

Closing this issue with: f82327c

closed 2023-02-28 by acxz