acxz/gazebo-arch #51: /usr/bin/gz exists in both 'ignition-tools' and 'gazebo'

petronny, 2023-03-03:
Which leads to some packages that depend on both of them fails to build.

https://github.com/arch4edu/cactus/actions/runs/4324667850/jobs/7549803120#step:15:977

muellerbernd, 2023-03-04:
I can confirm. With ignition-tools version 1.5.0 it builds without an error.

jannesklee, 2023-03-05:
I also have the error. What can I do in order to make the installation work. Should I downgrade to ignition-tools 1.5.0 or is there a more elegent way?

jannesklee, 2023-03-05:
I simply used --overwrite /usr/bin/gz. Its no clean solution but at the moment gazebo runs (I am a very beginner and simply want to try out some things)

acxz, 2023-03-06:
@petronny I believe I have resolved the issue, by adding a conflicts of `gazebo` on `ignition-tools`. I have also update several other packages to correctly refer to `ignition-tools-1` as needed.

Let me know if this resolves the issue.

petronny, 2023-03-07:
Thanks! Problem solved.

acxz, 2023-03-07:
Sweet! closing this issue for now.

closed 2023-03-07 by acxz