acxz/gazebo-arch #56: [ignition-common-3] AV_CODEC_CAP_TRUNCATED was not declared

acxz, 2023-03-11:
```
/home/acxz/vcs/git/github/acxz/gazebo-arch/ignition-common-3/src/gz-common-ignition-common3_3.15.1/av/src/AudioDecoder.cc: In member function ‘bool ignition::common::AudioDecoder::SetFile(const std::string&)’:
/home/acxz/vcs/git/github/acxz/gazebo-arch/ignition-common-3/src/gz-common-ignition-common3_3.15.1/av/src/AudioDecoder.cc:348:41: error: ‘AV_CODEC_CAP_TRUNCATED’ was not declared in this scope; did you mean ‘AV_CODEC_CAP_HARDWARE’?
  348 |   if (this->data->codec->capabilities & AV_CODEC_CAP_TRUNCATED)
      |                                         ^~~~~~~~~~~~~~~~~~~~~~
      |                                         AV_CODEC_CAP_HARDWARE
/home/acxz/vcs/git/github/acxz/gazebo-arch/ignition-common-3/src/gz-common-ignition-common3_3.15.1/av/src/AudioDecoder.cc:349:36: error: ‘AV_CODEC_FLAG_TRUNCATED’ was not declared in this scope; did you mean ‘AV_CODEC_FLAG_DROPCHANGED’?
  349 |     this->data->codecCtx->flags |= AV_CODEC_FLAG_TRUNCATED;
      |                                    ^~~~~~~~~~~~~~~~~~~~~~~
      |                                    AV_CODEC_FLAG_DROPCHANGED
make[2]: *** [av/src/CMakeFiles/ignition-common3-av.dir/build.make:76: av/src/CMakeFiles/ignition-common3-av.dir/AudioDecoder.cc.o] Error 1
```

acxz, 2023-03-11:
patched with a17771e

PSandro, 2023-03-21:
This error also occurs when building `ignition-commons`:
```make
home/sandro/.cache/yay/ignition-common/src/gz-common-gz-common5_5.3.1/av/src/Video.cc:199:29: error: ‘AV_CODEC_CAP_TRUNCATED’ was not declared in this scope; did you mean ‘AV_CODEC_CAP_HARDWARE’?
  199 |   if (codec->capabilities & AV_CODEC_CAP_TRUNCATED)
      |                             ^~~~~~~~~~~~~~~~~~~~~~
      |                             AV_CODEC_CAP_HARDWARE
/home/sandro/.cache/yay/ignition-common/src/gz-common-gz-common5_5.3.1/av/src/Video.cc:200:39: error: ‘AV_CODEC_FLAG_TRUNCATED’ was not declared in this scope; did you mean ‘AV_CODEC_FLAG_DROPCHANGED’?
  200 |     this->dataPtr->codecCtx->flags |= AV_CODEC_FLAG_TRUNCATED;
      |                                       ^~~~~~~~~~~~~~~~~~~~~~~
      |                                       AV_CODEC_FLAG_DROPCHANGED
```

acxz, 2023-03-21:
@PSandro thanks for the report, this should be resolved with [5a505f2](https://github.com/acxz/gazebo-arch/commit/5a505f2ff859986f98b72f47c30015e2e2c97168)