acxz/gazebo-arch #58: [ignition-tools] could not update ignition-tools due to ros-noetic-gazebo-dev

fbartelt, 2023-03-12:
While updating `ignition-tools=1.5.0-1` to `ignition-tools=2.0.0-2`:
```
:: ignition-tools and gazebo are in conflict. Remove gazebo? [y/N] y
error: failed to prepare transaction (could not satisfy dependencies)
:: removing gazebo breaks dependency 'gazebo' required by ros-noetic-gazebo-dev
``` 
How should this be handled? I'm using the `ros-noetic-desktop-full=1.5.0-1` metapackage
```
gazebo=11.12.0-2
ros-noetic-gazebo-dev=2.9.2-1
```

acxz, 2023-03-12:
The best way to handle this is to install `ignition-tools-1` to use with `gazebo`, instead of `ignition-tools`.

acxz, 2023-03-14:
If you are still having problems with the above issue, feel free to comment. Closing this issue as solved for now.

closed 2023-03-14 by acxz

fbartelt, 2023-03-15:
I could only test this today, sorry. Everything seems to be working now. Thanks!