acxz/gazebo-arch #61: 'd' was not declared in this scope.

jackleberryfin, 2023-03-15:
Target build : ignition-common-3 (3.15.1-2)

For me this issue is in addition to #56 and #57. 

I was able to continue with the build after removing these sections of preprocessor sections of code. 

[image]

[image]

jackleberryfin, 2023-03-15:
I'm a bit of a newb and its probably quite likely that this version i'm building is not the latest.. and has been patched already.. The problem is this version is a dependency in ros-noetic

muellerbernd, 2023-03-16:
@jackleberryfin ignition-common-3 is a dependency to gazebo and this is a dependency to ros-noetic-gazebo* . But at the moment the gazebo in AUR does not use the patched version. As soon as #54 is merged it should work.

acxz, 2023-03-17:
In the time being you can use `gazebo-git` instead of `gazebo`.

acxz, 2023-03-19:
`gazebo` has been updated, give it another shot @jackleberryfin

acxz, 2023-03-19:
Closing as dup of #56

also please make sure to post codeblocks rather than images so that we can search through code

closed 2023-03-19 by acxz