acxz/gazebo-arch #62: [sdformat-9] not building.

KavinTheG, 2023-03-16:
Target build: sdformat-9
```c++
/home/kavin/.cache/yay/sdformat-9/src/sdformat-sdformat9_9.8.0/src/parser_urdf.cc:30:10: fatal error: urdf_model/model.h: No such file or directory
   30 | #include <urdf_model/model.h>
      |          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [src/CMakeFiles/sdformat9.dir/build.make:552: src/CMakeFiles/sdformat9.dir/parser_urdf.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:3435: src/CMakeFiles/sdformat9.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
 -> error making: sdformat-9 
``` 
My issue is that when I try to install gazebo/gazebo-git, I am unable to build sdformat-9. 
I am utilising ros2-humble. I also have urdfdom and urdfdom-headers installed.

```
local/urdfdom 3.1.0-1
    The URDF (U-Robot Description Format) library provides core data structures and a simple XML parsers for populating the class data structures from an
    URDF file.
local/urdfdom-headers 1.1.0-1
    The URDF (U-Robot Description Format) headers provides core data structure headers for URDF.
```
Any help would be greatly appreciated!

KavinTheG, 2023-03-17:
It seemed that issue was mainly some sort of include error. I was able to resolve this by pulling the source code for sdformat at https://github.com/gazebosim/sdformat and replacing all occurances of '#include <urdf_*/*> with <urdf/urdf_*/*>.

bruno-anjos, 2023-04-08:
Package still fails building with the same error.

acxz, 2023-06-22:
@bruno-anjos I am not able to reproduce this error. Can you try building in clean chroot and report back?

acxz, 2023-09-26:
closing this error with no response and no reproducibility

codeflight1, 2023-12-22:
I have experienced the same error while installing `sdformat-9`.

acxz, 2023-12-22:
@codeflight1 were you able to try to build in a clean chroot?

codeflight1, 2024-01-01:
I attempted to, but I encountered missing dependency issues with the `ignition-*` packages even after installing them. I'm using `clean-chroot-manager`, not sure if that may have an impact.

acxz, 2024-01-01:
Yep that definitely sounds like a `clean-chroot-manager` usability error. If you want to build in a clean chroot, I'd suggest using `paru`. Please try again with `paru`.

codeflight1, 2024-01-02:
I was able to successfully build with `paru`, and I did not encounter the compilation error.

acxz, 2024-01-02:
Glad to hear! I'll go ahead and close this issue, if you have any more errors feel free to post.

closed 2024-01-02 by acxz