acxz/gazebo-arch #65: [ignition-common-3] Fails to build, error: 'uint32_t' has not been declared

Gnarflord, 2023-05-08:
The package currently fails to build correctly to me. Seems like the compilation unit is missing an `#include <stdint.h>`?

```
> yay -S ignition-common-3
[...]
[ 98%] Building CXX object profiler/src/CMakeFiles/ignition-common3-profiler.dir/RemoteryProfilerImpl.cc.o
In file included from /home/jan/.cache/yay/ignition-common-3/src/gz-common-ignition-common3_3.15.1/profiler/src/RemoteryProfilerImpl.hh:26,
                 from /home/jan/.cache/yay/ignition-common-3/src/gz-common-ignition-common3_3.15.1/profiler/src/RemoteryProfilerImpl.cc:20:
/home/jan/.cache/yay/ignition-common-3/src/gz-common-ignition-common3_3.15.1/profiler/src/ProfilerImpl.hh:56:59: error: ‘uint32_t’ has not been declared
   56 |       public: virtual void BeginSample(const char *_name, uint32_t *_hash) = 0;
      |                                                           ^~~~~~~~
/home/jan/.cache/yay/ignition-common-3/src/gz-common-ignition-common3_3.15.1/profiler/src/RemoteryProfilerImpl.hh:74:51: error: ‘uint32_t’ has not been declared
   74 |       public: void BeginSample(const char *_name, uint32_t *_hash) final;
      |                                                   ^~~~~~~~
/home/jan/.cache/yay/ignition-common-3/src/gz-common-ignition-common3_3.15.1/profiler/src/RemoteryProfilerImpl.cc:226:6: error: no declaration matches ‘void ignition::common::RemoteryProfilerImpl::BeginSample(const char*, uint32_t*)’
  226 | void RemoteryProfilerImpl::BeginSample(const char *_name, uint32_t *_hash)
      |      ^~~~~~~~~~~~~~~~~~~~
/home/jan/.cache/yay/ignition-common-3/src/gz-common-ignition-common3_3.15.1/profiler/src/RemoteryProfilerImpl.hh:74:20: note: candidate is: ‘virtual void ignition::common::RemoteryProfilerImpl::BeginSample(const char*, int*)’
   74 |       public: void BeginSample(const char *_name, uint32_t *_hash) final;
      |                    ^~~~~~~~~~~
/home/jan/.cache/yay/ignition-common-3/src/gz-common-ignition-common3_3.15.1/profiler/src/RemoteryProfilerImpl.hh:47:11: note: ‘class ignition::common::RemoteryProfilerImpl’ defined here
   47 |     class RemoteryProfilerImpl: public ProfilerImpl
      |           ^~~~~~~~~~~~~~~~~~~~
make[2]: *** [profiler/src/CMakeFiles/ignition-common3-profiler.dir/build.make:104: profiler/src/CMakeFiles/ignition-common3-profiler.dir/RemoteryProfilerImpl.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:648: profiler/src/CMakeFiles/ignition-common3-profiler.dir/all] Error 2
make: *** [Makefile:166: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
 -> error making: ignition-common-3-exit status 4
 -> Failed to install the following packages. Manual intervention is required:
ignition-common-3 - exit status 4
```

acxz, 2023-06-22:
Closed with [1828bc5](https://github.com/acxz/gazebo-arch/commit/1828bc5c5c0de58887ca00b3c7ef34fbb6316ad1)

lounick, 2023-07-03:
This is also affecting the v5 build of common. Has been patched upstream but without a new release. Could we possibly cherrypick that and apply that without too much hassle?