acxz/gazebo-arch #68: [ignition-msgs-5] compile error DSO missing from command line

muellerbernd, 2023-07-19:
While rebuilding gazebo classic packages I get the following error:
```bash
/sbin/ld: CMakeFiles/ign_msgs_gen.dir/Generator.cc.o: undefined reference to symbol '_ZN4absl12lts_2023012513hash_internal15MixingHashState28CombineLargeContiguousImpl64EmPKhm'
/sbin/ld: /usr/lib/libabsl_hash.so.2301.0.0: error adding symbols: DSO missing from command line
```
This error occurs while building _ignition-mgs-5_ and _gazebo_

petronny, 2023-08-01:
Can be confirmed in https://github.com/arch4edu/cactus/actions/runs/5721641047/job/15503542368

acxz, 2023-09-26:
patched with #76 
listed as patched, since I believe those flags should be part of the build configuration in cmake rather than something we keep track of in the PKGBUILD. Since we are keeping track of it in the PKGBUILD, its important to make sure we know why those flags are needed.

acxz, 2023-09-27:
We keep issues that have been patched open, until the patch is no longer needed.