acxz/gazebo-arch #70: [ignition-transport-8] compile error protobuf

muellerbernd, 2023-07-19:
After protobuf update the package does not build anymore

petronny, 2023-09-18:
Build log:
```
In file included from /build/ignition-transport-8/src/gz-transport-ignition-transport8_8.4.0/include/gz/transport/NodeShared.hh:41,
                 from /build/ignition-transport-8/src/gz-transport-ignition-transport8_8.4.0/include/gz/transport/Node.hh:42,
                 from /build/ignition-transport-8/src/gz-transport-ignition-transport8_8.4.0/src/CIface.cc:21:
/build/ignition-transport-8/src/gz-transport-ignition-transport8_8.4.0/include/gz/transport/RepHandler.hh:30:10: fatal error: google/protobuf/stubs/casts.h: No such file or directory
   30 | #include <google/protobuf/stubs/casts.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [src/CMakeFiles/ignition-transport8.dir/build.make:90: src/CMakeFiles/ignition-transport8.dir/CIface.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:346: src/CMakeFiles/ignition-transport8.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[  8%] Built target doc
make: *** [Makefile:146: all] Error 2
==> ERROR: A failure occurred in build().
```
https://github.com/arch4edu/cactus/actions/runs/6219832833/job/16878662591

acxz, 2023-09-26:
patched with #73