acxz/gazebo-arch #80: [gazebo] Symbol look up error

fromtheeast710, 2023-08-05:
I'm having trouble running `gazebo`, which fails to start and return:

```
gazebo-11.12.0: symbol lookup error: /usr/lib/libignition-msgs5.so.5: undefined symbol: _ZN6google8protobuf13RepeatedFieldIdED1Ev
```

The package I'm using is `ros-noetic-desktop-full 1.5.0-2` from the AUR. 

The default installation use `libprotobuf.so.23` but that was not available on my machine so I symlinked it from `/usr/lib/libprotobuf.so.23.4.0`

realSaddy, 2023-08-05:
Try checking out the commit from [this PR](https://github.com/acxz/gazebo-arch/pull/73) and using `makepkg -si`. Fixed this issue for me yesterday.

fromtheeast710, 2023-08-06:
@realSaddy Thank you and that did solve the problem with `libignition-msgs-5` but then gazebo showed this error:

```
gazebo: symbol lookup error: /usr/lib/libignition-fuel_tools4.so.4: undefined symbol: _ZN8ignition4msgs12FuelMetadataC1EPN6google8protobuf5ArenaEb
```

So I recompiled `ignition-fuel_tools-4` too and that led to `ignition-transport-8` showing that error message. However, I was unable to recompile that package.

Here's the full log:

```
$ mkpkg -si
==> WARNING: Cannot find the sudo binary. Will use su to acquire root privileges.
==> Making package: ignition-transport-8 8.4.0-2 (Sun 06 Aug 2023 07:10:02 PM AEST)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Found ignition-transport8_8.4.0.tar.gz
==> Validating source files with sha256sums...
    ignition-transport8_8.4.0.tar.gz ... Passed
==> Extracting sources...
  -> Extracting ignition-transport8_8.4.0.tar.gz with bsdtar
==> Removing existing $pkgdir/ directory...
==> Starting build()...
-- ignition-transport8 version 8.4.0
-- Operating system is Linux
-- 

-- ====== Finding Dependencies ======
-- Looking for Protobuf - found

-- Looking for ZeroMQ - found

CMake Warning at CMakeLists.txt:59 (message):
  lsb_release executable not found.  Disabling focal-specific workarounds

-- Looking for CppZMQ - not found

-- Checking for module 'uuid'
--   Found uuid, version 2.39.1
-- Looking for UUID - found

-- Looking for ignition-msgs5 -- found version 5.11.0
-- Searching for dependencies of ignition-msgs5
-- Looking for ignition-math6 -- found version 6.14.0
-- Searching for dependencies of ignition-math6
-- Checking for module 'tinyxml2'
--   Found tinyxml2, version 9.0.0
-- Looking for ignition-msgs5 - found

-- Looking for IFADDRS - found

-- Checking for module 'sqlite3 >= 3.7.13'
--   Found sqlite3 , version 3.42.0
-- Looking for sqlite3 - found

CMake Error at /usr/share/cmake/ignition-cmake2/cmake2/IgnConfigureBuild.cmake:59 (message):
  -- BUILD ERRORS: These must be resolved before compiling.
Call Stack (most recent call first):
  CMakeLists.txt:112 (ign_configure_build)

CMake Error at /usr/share/cmake/ignition-cmake2/cmake2/IgnConfigureBuild.cmake:61 (message):
  -- 	Missing dependency [CppZMQ]
Call Stack (most recent call first):
  CMakeLists.txt:112 (ign_configure_build)

CMake Error at /usr/share/cmake/ignition-cmake2/cmake2/IgnConfigureBuild.cmake:63 (message):
  -- END BUILD ERRORS

Call Stack (most recent call first):
  CMakeLists.txt:112 (ign_configure_build)

CMake Error at /usr/share/cmake/ignition-cmake2/cmake2/IgnConfigureBuild.cmake:68 (message):
  Errors encountered in build.  Please see BUILD ERRORS above.
Call Stack (most recent call first):
  CMakeLists.txt:112 (ign_configure_build)

-- Configuring incomplete, errors occurred!
==> ERROR: A failure occurred in build().
    Aborting...
```

realSaddy, 2023-08-06:
@fromtheeast710 Check out [this PR](https://github.com/acxz/gazebo-arch/pull/73) and build ignition-transport-8. 
@acxz any word on when these prs will be merged? arch upstream is broken currently because of these issues

fromtheeast710, 2023-08-12:
@realSaddy Finally got the time to get around this. I have recompiled the package but now I'm stuck at

```
gzserver: symbol lookup error: /usr/lib/libgazebo_msgs.so.11: undefined symbol: _ZN6google8protobuf13RepeatedFieldIjED1Ev
```

What should I do next?

Thank you very much for the help.

@acxz Can we please merge the PRs?

realSaddy, 2023-08-12:
@fromtheeast710 Looks like you updated protobuf since the last build of gazebo_msgs. Try rebuilding gazebo_msgs

fromtheeast710, 2023-08-13:
@realSaddy I have rebuilt `ros-noetic-gazebo-msgs` but the problem still persist.

fromtheeast710, 2023-08-27:
@realSaddy Sorry for keep mentioning you but I'm really stuck here. I have tried everything including rebuilding all packages and gazebo but gazebo failed to compile and I'm stuck at:

```
gzserver: symbol lookup error: /usr/lib/libgazebo_msgs.so.11: undefined symbol: _ZN6google8protobuf13RepeatedFieldIjED1Ev
```
How did you get it working on your machine and what do you mean by my protobuf was updated? Originally I used the packages from `arch4edu` because the ones from `AUR` failed to compile.

I just want to get my project started ;_;

realSaddy, 2023-08-27:
@fromtheeast710 Np. Can you try checking out [this PR](https://github.com/acxz/gazebo-arch/pull/76) and running `sudo makepkg -si` inside of ignition-msgs-5?

fromtheeast710, 2023-08-29:
That didn't fix the problem sadly. When I did:

```
$ yay -Qo /usr/lib/libgazebo_msgs.so.11
/usr/lib/libgazebo_msgs.so.11 is owned by gazebo 11.12.0-7
```

So I think the problem is with gazebo itself but gazebo failed to compile on my machine with error:

```
[ 51%] Building CXX object gazebo/physics/CMakeFiles/gazebo_physics.dir/BoxShape.cc.o
In file included from /home/theeast/Projects/gazebo-arch/gazebo/src/gazebo-classic-gazebo11_11.13.0/gazebo/rendering/deferred_shading/GBufferMaterialGenerator.hh:20,
                 from /home/theeast/Projects/gazebo-arch/gazebo/src/gazebo-classic-gazebo11_11.13.0/gazebo/rendering/deferred_shading/GBufferMaterialGenerator.cc:27:
/home/theeast/Projects/gazebo-arch/gazebo/src/gazebo-classic-gazebo11_11.13.0/gazebo/rendering/deferred_shading/MaterialGenerator.hh:42:23: error: ‘uint32_t’ does not name a type
   42 |       public: typedef uint32_t Perm;
      |                       ^~~~~~~~
/home/theeast/Projects/gazebo-arch/gazebo/src/gazebo-classic-gazebo11_11.13.0/gazebo/rendering/deferred_shading/MaterialGenerator.hh:26:1: note: ‘uint32_t’ is defined in header ‘<cstdint>’; did you forget to ‘#include <cstdint>’?
   25 | #include "gazebo/util/system.hh"
  +++ |+#include <cstdint>
   26 | 
/home/theeast/Projects/gazebo-arch/gazebo/src/gazebo-classic-gazebo11_11.13.0/gazebo/rendering/deferred_shading/MaterialGenerator.hh:52:52: error: ‘Perm’ has not been declared
   52 |       public: const Ogre::MaterialPtr &GetMaterial(Perm permutation);
      |                                                    ^~~~
/home/theeast/Projects/gazebo-arch/gazebo/src/gazebo-classic-gazebo11_11.13.0/gazebo/rendering/deferred_shading/MaterialGenerator.hh:63:29: error: ‘Perm’ has not been declared
   63 |                             Perm _permutation) = 0;
      |                             ^~~~
/home/theeast/Projects/gazebo-arch/gazebo/src/gazebo-classic-gazebo11_11.13.0/gazebo/rendering/deferred_shading/MaterialGenerator.hh:65:29: error: ‘Perm’ has not been declared
   65 |                             Perm _permutation) = 0;
      |                             ^~~~
/home/theeast/Projects/gazebo-arch/gazebo/src/gazebo-classic-gazebo11_11.13.0/gazebo/rendering/deferred_shading/MaterialGenerator.hh:67:29: error: ‘Perm’ has not been declared
   67 |                             Perm permutation) = 0;
      |                             ^~~~
/home/theeast/Projects/gazebo-arch/gazebo/src/gazebo-classic-gazebo11_11.13.0/gazebo/rendering/deferred_shading/MaterialGenerator.hh:71:61: error: ‘Perm’ has not been declared
   71 |       protected: const Ogre::GpuProgramPtr &GetVertexShader(Perm _perm);
      |                                                             ^~~~
/home/theeast/Projects/gazebo-arch/gazebo/src/gazebo-classic-gazebo11_11.13.0/gazebo/rendering/deferred_shading/MaterialGenerator.hh:72:63: error: ‘Perm’ has not been declared
   72 |       protected: const Ogre::GpuProgramPtr &GetFragmentShader(Perm _perm);
      |                                                               ^~~~
/home/theeast/Projects/gazebo-arch/gazebo/src/gazebo-classic-gazebo11_11.13.0/gazebo/rendering/deferred_shading/MaterialGenerator.hh:73:63: error: ‘Perm’ has not been declared
   73 |       protected: const Ogre::MaterialPtr &GetTemplateMaterial(Perm _perm);
      |                                                               ^~~~
/home/theeast/Projects/gazebo-arch/gazebo/src/gazebo-classic-gazebo11_11.13.0/gazebo/rendering/deferred_shading/MaterialGenerator.hh:81:18: error: ‘Perm’ does not name a type
   81 |       protected: Perm vsMask;
      |                  ^~~~
/home/theeast/Projects/gazebo-arch/gazebo/src/gazebo-classic-gazebo11_11.13.0/gazebo/rendering/deferred_shading/MaterialGenerator.hh:84:18: error: ‘Perm’ does not name a type
   84 |       protected: Perm fsMask;
      |                  ^~~~
/home/theeast/Projects/gazebo-arch/gazebo/src/gazebo-classic-gazebo11_11.13.0/gazebo/rendering/deferred_shading/MaterialGenerator.hh:87:18: error: ‘Perm’ does not name a type
   87 |       protected: Perm matMask;
      |                  ^~~~
/home/theeast/Projects/gazebo-arch/gazebo/src/gazebo-classic-gazebo11_11.13.0/gazebo/rendering/deferred_shading/MaterialGenerator.hh:92:35: error: ‘Perm’ was not declared in this scope
   92 |       protected: typedef std::map<Perm, Ogre::GpuProgramPtr> ProgramMap;
      |                                   ^~~~
/home/theeast/Projects/gazebo-arch/gazebo/src/gazebo-classic-gazebo11_11.13.0/gazebo/rendering/deferred_shading/MaterialGenerator.hh:92:60: error: template argument 1 is invalid
   92 |       protected: typedef std::map<Perm, Ogre::GpuProgramPtr> ProgramMap;
      |                                                            ^
/home/theeast/Projects/gazebo-arch/gazebo/src/gazebo-classic-gazebo11_11.13.0/gazebo/rendering/deferred_shading/MaterialGenerator.hh:92:60: error: template argument 3 is invalid
/home/theeast/Projects/gazebo-arch/gazebo/src/gazebo-classic-gazebo11_11.13.0/gazebo/rendering/deferred_shading/MaterialGenerator.hh:92:60: error: template argument 4 is invalid
/home/theeast/Projects/gazebo-arch/gazebo/src/gazebo-classic-gazebo11_11.13.0/gazebo/rendering/deferred_shading/MaterialGenerator.hh:93:35: error: ‘Perm’ was not declared in this scope
   93 |       protected: typedef std::map<Perm, Ogre::MaterialPtr> MaterialMap;
      |                                   ^~~~
[... cut ...]
  302 |     MaterialGenerator::Perm _permutation)
      |                        ^~~~
/home/theeast/Projects/gazebo-arch/gazebo/src/gazebo-classic-gazebo11_11.13.0/gazebo/rendering/deferred_shading/GBufferMaterialGenerator.cc:303:1: error: expected ‘,’ or ‘;’ before ‘{’ token
  303 | {
      | ^
make[2]: *** [gazebo/rendering/CMakeFiles/gazebo_rendering.dir/build.make:832: gazebo/rendering/CMakeFiles/gazebo_rendering.dir/deferred_shading/GBufferMaterialGenerator.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
```

acxz, 2023-09-26:
Truly apologize for my lack of attention to this repository folks. The associated PRs have been merged and should unblock this issue. If you are still have problems, please make another issue and we can take it from there.

closed 2023-09-26 by acxz