acxz/gazebo-arch #83: [gazebo] MaterialGenerator.hh:42:23: error: ‘uint32_t’ does not name a type

petronny, 2023-09-27:
```
[ 54%] Building CXX object gazebo/rendering/CMakeFiles/gazebo_rendering.dir/deferred_shading/DeferredLight.cc.o
[ 54%] Building CXX object gazebo/physics/CMakeFiles/gazebo_physics.dir/Road.cc.o
[ 54%] Building CXX object gazebo/rendering/CMakeFiles/gazebo_rendering.dir/deferred_shading/GBufferMaterialGenerator.cc.o
In file included from /build/gazebo/src/gazebo-classic-gazebo11_11.13.0/gazebo/rendering/deferred_shading/GBufferMaterialGenerator.hh:20,
                 from /build/gazebo/src/gazebo-classic-gazebo11_11.13.0/gazebo/rendering/deferred_shading/GBufferMaterialGenerator.cc:27:
/build/gazebo/src/gazebo-classic-gazebo11_11.13.0/gazebo/rendering/deferred_shading/MaterialGenerator.hh:42:23: error: ‘uint32_t’ does not name a type
   42 |       public: typedef uint32_t Perm;
      |                       ^~~~~~~~
/build/gazebo/src/gazebo-classic-gazebo11_11.13.0/gazebo/rendering/deferred_shading/MaterialGenerator.hh:26:1: note: ‘uint32_t’ is defined in header ‘<cstdint>’; did you forget to ‘#include <cstdint>’?
   25 | #include "gazebo/util/system.hh"
```
https://github.com/arch4edu/cactus/actions/runs/6315323603/job/17167775524

muellerbernd, 2023-09-27:
I think this comes from VTK. I had a similar issues while using the extra package. Then I tried to build it locally and it failed with a similar error. Now I use a patched version of VTK that uses [upstream patches](https://github.com/Kitware/VTK/commit/1233ceec) you can find it [here](https://github.com/muellerbernd/my_arch_packages/blob/master/vtk-bemu/PKGBUILD).
I think VTK needs a rebuild.

acxz, 2023-12-17:
After some gazebo updates look this error is resolved: https://github.com/arch4edu/cactus/actions/runs/7189621455, closing for now.
Feel free to comment if this pops up again.

closed 2023-12-17 by acxz