acxz/gazebo-arch #88: [gazebo] Installation error

Reversean, 2023-11-26:
Log:

```
/bin/sh: line 1: --plugin=protoc-gen-gazebomsgs=/home/Reversean/.cache/yay/gazebo/src/gazebo-classic-gazebo11_11.14.0/build/gazebo/msgs/gazebomsgs_out: No such file or directory
make[2]: *** [gazebo/msgs/CMakeFiles/gazebo_msgs.dir/build.make:1107: gazebo/msgs/wrench_stamped.pb.cc] Error 127
make[1]: *** [CMakeFiles/Makefile2:2483: gazebo/msgs/CMakeFiles/gazebo_msgs.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
 -> error making: gazebo-exit status 4
 -> Failed to install the following packages. Manual intervention is required:
gazebo - exit status 4
```

OS: Arch Linux x86_64

Kernel: 6.6.2-arch1-1

Reversean, 2023-11-26:
I looked in the problem file (`/home/Reversean/.cache/yay/gazebo/src/gazebo-classic-gazebo11_11.14.0/build/gazebo/msgs/CMakeFiles/gazebo_msgs.dir/build.make`):

```
...

gazebo/msgs/wrench_stamped.pb.cc: /home/Reversean/.cache/yay/gazebo/src/gazebo-classic-gazebo11_11.14.0/gazebo/msgs/wrench_stamped.proto
gazebo/msgs/wrench_stamped.pb.cc: gazebo/msgs/gazebomsgs_out
        @$(CMAKE_COMMAND) -E cmake_echo_color "--switch=$(COLOR)" --blue --bold --progress-dir=/home/Reversean/.cache/yay/gazebo/src/gazebo-classic-gazebo11_11.14.0/build/CMakeFiles --progress-num=$(CMAKE_PROGRESS_130) "Running C++ protocol buffer compiler on wrench_stamped.proto"
        cd /home/Reversean/.cache/yay/gazebo/src/gazebo-classic-gazebo11_11.14.0/build/gazebo/msgs && --plugin=protoc-gen-gazebomsgs=/home/Reversean/.cache/yay/gazebo/src/gazebo-classic-gazebo11_11.14.0/build/gazebo/msgs/gazebomsgs_out --cpp_out=dllexport_decl=GZ_MSGS_VISIBLE:/home/Reversean/.cache/yay/gazebo/src/gazebo-classic-gazebo11_11.14.0/build/gazebo/msgs --gazebomsgs_out=/home/Reversean/.cache/yay/gazebo/src/gazebo-classic-gazebo11_11.14.0/build/gazebo/msgs --proto_path=/home/Reversean/.cache/yay/gazebo/src/gazebo-classic-gazebo11_11.14.0/gazebo/msgs /home/Reversean/.cache/yay/gazebo/src/gazebo-classic-gazebo11_11.14.0/gazebo/msgs/wrench_stamped.proto

...
```

Seems to be missing a command `protoc` before `--plugin=...`, but can't understand why...

kvkoskin, 2023-12-08:
I have the same issue. `PROTOBUF_PROTOC_EXECUTABLE` is not set for some reason. I edited the cmake invocation in `PKGBUILD`, just hardcoding the path.
```
cmake .. -DCMAKE_BUILD_TYPE="Release" \
         -DCMAKE_INSTALL_PREFIX="/usr" \
         -DCMAKE_INSTALL_LIBDIR="lib" \
         -DPROTOBUF_PROTOC_EXECUTABLE="/usr/bin/protoc"
```
Seems to work around the real issue. At least gazebo compiles now.

acxz, 2023-12-11:
patched with [a138f78](https://github.com/acxz/gazebo-arch/commit/a138f788a6408be0fadda1cda0ffc404dfb8bc43)