acxz/gazebo-arch #91: [sdformat] Error while building

piratecarrot, 2024-02-18:
`build/src/CMakeFiles/UNIT_parser_urdf_TEST.dir/flags.make:CXX_FLAGS = -march=x86-64 -mtune=generic -O2 -pipe -fno-plt -fexceptions         -Wp,-D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security         -fstack-clash-protection -fcf-protection -Wp,-D_GLIBCXX_ASSERTIONS -g -ffile-prefix-map=/home/xxxxx/.cache/yay/sdformat/src=/usr/src/debug/sdformat -flto=auto -O3 -DNDEBUG  -Wall -Wextra -Wno-long-long -Wno-unused-value -Wfloat-equal -Wshadow -Winit-self -Wswitch-default -Wmissing-include-dirs -pedantic  -std=c++17 -fvisibility=hidden -msse -msse2 -mfpmath=sse -msse3 -mssse3 -msse4.1 -msse4.2 -I/usr/include -I/opt/ros/humble/lib/pkgconfig/../../include/urdfdom -I/home/xxxxx/.cache/yay/ros2-humble/src/install/include/urdfdom_headers -I/home/xxxxx/.cache/yay/ros2-humble/src/build/console_bridge_vendor/console_bridge_vendor_install/include/console_bridge_vendor`

Bizarre include directory, it is assuming that ros2-humble build location still exists, which it may or may not. In my case, it does not.

piratecarrot, 2024-02-18:
I should note that I am using yay AUR manager, if that wasn't immediatebly obvious.

BBArikL, 2024-08-27:
Still cannot build on Arch Llinux, building fails with:

```
... (compilation done by yay)
[ 34%] Building CXX object test/integration/CMakeFiles/INTEGRATION_element_memory_leak.dir/element_memory_leak.cc.o
[ 34%] Linking CXX executable INTEGRATION_element_memory_leak
[ 34%] Built target INTEGRATION_element_memory_leak
[ 34%] Running xml schema compiler on actor.sdf
/home/<user>/.cache/yay/sdformat-9/src/sdformat-sdformat9_9.8.0/tools/xmlschema.rb:268:in `<main>': undefined method `exists?' for File:Class (NoMethodError)

elsif !File.exists?(infile)
           ^^^^^^^^
Did you mean?  exist?
make[2]: *** [sdf/1.7/CMakeFiles/schema1_7.dir/build.make:127: sdf/1.7/actor.xsd] Error 1
make[1]: *** [CMakeFiles/Makefile2:3565: sdf/1.7/CMakeFiles/schema1_7.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
 -> Failed to install layer, rolling up to next layer.error:error making: sdformat-9 - exit status 4
```

marco-perin, 2024-10-17:
I cannot build as well with the same error as @BBArikL. I am using yay as AUR manager as well

vladh, 2024-12-03:
Same error here. `File.exists?` was deprecated in Ruby 2.2 and removed in Ruby 3.2.0. A downgrade would fix it, but downgrading Ruby would probably break other stuff. So `stdformat` probably needs a fix.

vladh, 2024-12-03:
I fixed this by going to `sdformat-9/src/sdformat-sdformat9_9.8.0/tools/xmlschema.rb` and running the substitution `s/exists?/exist?`.

biohazardpb4, 2025-01-14:
https://github.com/acxz/gazebo-arch/pull/107 is a potential fix