acxz/gazebo-arch #94: [gazebo] Gazebo fails to compile

prximenes, 2024-02-21:
```[ 58%] Building CXX object gazebo/gui/CMakeFiles/gazebo_gui.dir/qgv/QGVSubGraph.cpp.o
/home/pedro/applications/gazebo/src/gazebo-classic-gazebo11_11.14.0/gazebo/gui/qgv/QGVSubGraph.cpp: In member function ‘QGVNode* QGVSubGraph::addNode(const QString&)’:
/home/pedro/applications/gazebo/src/gazebo-classic-gazebo11_11.14.0/gazebo/gui/qgv/QGVSubGraph.cpp:47:53: error: ‘TRUE’ was not declared in this scope
   47 |     Agnode_t *node = agnode(_sgraph->graph(), NULL, TRUE);
      |                                                     ^~~~
/home/pedro/applications/gazebo/src/gazebo-classic-gazebo11_11.14.0/gazebo/gui/qgv/QGVSubGraph.cpp: In member function ‘QGVSubGraph* QGVSubGraph::addSubGraph(const QString&, bool)’:
/home/pedro/applications/gazebo/src/gazebo-classic-gazebo11_11.14.0/gazebo/gui/qgv/QGVSubGraph.cpp:69:56: error: ‘TRUE’ was not declared in this scope
   69 |             ("cluster_" + _name).toLocal8Bit().data(), TRUE);
      |                                                        ^~~~
/home/pedro/applications/gazebo/src/gazebo-classic-gazebo11_11.14.0/gazebo/gui/qgv/QGVSubGraph.cpp:72:71: error: ‘TRUE’ was not declared in this scope
   72 |    sgraph = agsubg(_sgraph->graph(), _name.toLocal8Bit().data(), TRUE);
      |                                                                  ^~~~

make[2]: *** [gazebo/gui/CMakeFiles/gazebo_gui.dir/build.make:1852: gazebo/gui/CMakeFiles/gazebo_gui.dir/qgv/QGVSubGraph.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:3818: gazebo/gui/CMakeFiles/gazebo_gui.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
==> ERROR: A failure occurred in build().```

prximenes, 2024-02-21:
Hello,

I've found a workaround for this error. The issue is related to the version of Graphviz being used. Below are the detailed steps I followed to resolve the issue:

I started by removing the version of graphviz that was currently installed on my system. I did this by executing the following command in the terminal:
```
sudo pacman -Rns graphviz
```
Then, I cloned the graphviz version 9 package repository from Arch Linux's GitLab. For this, I used the following command:
```
git clone https://gitlab.archlinux.org/archlinux/packaging/packages/graphviz.git
```
Inside the cloned repository directory, I checked out to the specific 9.0.0-1 version of graphviz by executing:
```
git checkout eabf65842071603878f55c7e5d7bedfdf8144658
```
With the correct version of the graphviz source code in the directory, I proceeded with compilation and installation using makepkg with the command:
```
makepkg -si
```
After successfully installing the correct version of graphviz, I attempted to install Gazebo again through the AUR using  makepkg -si command. This time, the installation process completed without the previous compilation errors.

traversaro, 2024-02-21:
See https://github.com/gazebosim/gazebo-classic/pull/3373 .

petronny, 2024-03-29:
Also getting this error. Please update the PKGBUILD to include this patch.
Full build log: 
https://github.com/arch4edu/cactus/actions/runs/8475957989/job/23241002816