acxz/gazebo-arch #101: Error build ignition-transport8

mrxiuer, 2024-05-27:
When I install ignition-transport8, it reports an error: [image] Does anyone know how to fix it?

clq9920, 2024-05-29:
i also encountered this problem ：<
maybe you can download the package from https://mirrors.tuna.tsinghua.edu.cn/arch4edu/x86_64/ignition-transport-8-8.4.0-4-x86_64.pkg.tar.zst

mrxiuer, 2024-05-29:
> 我也遇到了这个问题：<也许你可以从https://mirrors.tuna.tsinghua.edu.cn/arch4edu/x86_64/ignition-transport-8-8.4.0-4-x86_64.pkg.tar.zst 下载软件包[](https://mirrors.tuna.tsinghua.edu.cn/arch4edu/x86_64/ignition-transport-8-8.4.0-4-x86_64.pkg.tar.zst)

Thanks,it works.

closed 2024-05-29 by mrxiuer